MissouriMRDT/Autonomy_Software_Python #7: Investigate differential gps for improved accuracy

omc8db, 2016-10-24:


omc8db, 2016-10-29:
Did a few measurements of GPS drift today:

[image]

[image]

It's not just a Gaussian noise distribution like I had expected- the drift is smoothly progressing. Right now my theory is that the drift is caused by factors outside the GPS, like satellite movement and atmospheric conditions. For two identical receivers in nearby locations, I think the drift progression will look very similar. 

To test this, we need to get two identical NEO-6M GPS units, place them about 20 meters apart in a parking lot, and record the drift functions. Run a statistical correlation on the data sets. If the statistical correlation is greater than about 0.8, then it means that my hunch was right.

If my hunch was right, then we can put a GPS unit on the base station (with a known fixed GPS location), calculate the drift from that, the subtract the drift from the on-rover measurements to substantially improve accuracy.

cjh7z4, 2017-10-07:
https://www.google.com/patents/US5323322 This has a good explanation of how the gps system works and how differential gps works. So basically we use a GPS at a know location, the base station and a GPS on the Rover. We look at errors in the position of the base station and we correct for the same errors in the position of the rover. since they are close, the error should also be close. To me its kind of like a system response h(t). the system being the error generated by satellite position, ionospheric and tropospheric wave-front delay and errors in wave measurement. h(t) can be found by looking at differences between know and gps generated position and then this h(t) can be used with the gps position on rover to correct for better positioning. so In my mind there are several problems to work.

How to get an accurate position at base station

How to write code to characterize h(t) and how to send this to rover? Is this allowed during autonomous portion? it will involve communication over the network?

cjh7z4, 2017-10-08:
So after talking to Judah, I think a first step is to put the navboard software on a couple of TivaCs that have the navboard attachment and look at the data gathered from those board as they sit side by side and look to see if they produce similar data. Worked a bit on getting navboard software flashed to a tiva and serial data sent back to my laptop with Energia. I was unsuccessful in this endeavor and will not be able to pick it back up probably till late next week. Just posting this to give anyone who wants to help a starting point on what needs done. Basically we need this data to prove that locational drift is a function of the atmosphere and satellite position and not problems with our equipment.  If we can get similar gps data from 2 of our navboards in nearly the same location then we know that this may be worth pursuing at least with our current equipment. Please feel free to comment with suggestions, corrections, questions or progress made on this issue.

closed 2020-08-09 by EliVerbrugge