MissouriMRDT/Autonomy_Software_Python #8: Implement Crosstrack Correction

omc8db, 2016-10-25:
When approaching a waypoint, steer around it a bit so that when you pass through you're already pointing toward the next one.

omc8db, 2016-11-15:
Resources:

http://gis.stackexchange.com/questions/121117/cross-track-distance-to-great-cicle-line-and-rhumb-line
http://geo.javawa.nl/coordcalc/index_en.html - Look at calculation number 2

We're looking to calculate the distance and heading to the line between two waypoints

kohariker, 2016-11-15:
Here is an article that basically explains exactly how to find our "D" value on original course (even gives a javascript version of the algorithm). Now we just need to decide how far we want to correct.
http://www.movable-type.co.uk/scripts/latlong.html

I have not read their code, just the cross track logic.

closed 2017-04-09 by omc8db