MissouriMRDT/Autonomy_Software_Python #15: Environment Mapping

omc8db, 2017-04-09:
Build up a 3-d map of the environment around the rover with [rtabmap](http://wiki.ros.org/rtabmap_ros/Tutorials/StereoOutdoorMapping) to avoid obstacles. Depends on #10

omc8db, 2017-04-11:
Pull the video logs from the rover and run them through rtabmap on the ground

omc8db, 2017-09-15:
Techniques to investigate: 
* Stereoscopic Vision
* Structure from Motion
* LIDAR
* Deep Learning
* Some combination of the above

omc8db, 2017-10-07:
General Purpose Point Cloud Mapping http://pointclouds.org

captainabedabes, 2017-11-04:
We can successfully run the rtabmap tutorial linked above on a laptop running Ubuntu.
The next step is to get ROS running on the rover (likely on a Pi) and set up as detailed [here](http://wiki.ros.org/rtabmap_ros/Tutorials/SetupOnYourRobot).

sdlouhy, 2018-01-26:
Turns out most rtabmap-related issues have been coming from the fact that the odometry dependency (openni) has been [disabled](https://github.com/ros-drivers/openni_camera/issues/66) due to bugs on several distributions of ros.

sdlouhy, 2018-01-27:
[This](http://official-rtab-map-forum.67519.x6.nabble.com/template/NamlServlet.jtp?macro=user_nodes&user=1) is also a great forum to look at whenever you have any issues with rtabmap.

sdlouhy, 2018-02-01:
Forgot to add the [camera calibration](http://wiki.ros.org/camera_calibration/Tutorials/StereoCalibration) link.  The one difficult thing about this is making sure the cameras have the same timestamp

130s, 2018-09-13:
I'm not following this ticket at all but just FYI a ticket https://github.com/ros-drivers/openni_camera/issues/66 referenced in https://github.com/MissouriMRDT/AutonomyBoard_Software/issues/15#issuecomment-360893706 is now closed.

sdlouhy, 2018-09-14:
Environment mapping plans have changed a little in the past year.  The current plan is to use [PL_SLAM](https://github.com/rubengooj/pl-slam) for environment mapping, and a combination of stereoscopic cameras and a 1-D LiDAR for obstacle avoidance.

closed 2020-08-09 by EliVerbrugge