MissouriMRDT/Autonomy_Software_Python #105: Clean up math by precomputing the conditionals

declan34, 2021-08-13:
Maybe consider pre-computing and naming the result of these conditionals (which are used multiple times). For example, calling this heading = "north_south" or something like that may make the code a lot more legible.

_Originally posted by @thomas-gleiforst in https://github.com/MissouriMRDT/Autonomy_Software/pull/102#discussion_r644402932_

Byrdman32, 2022-09-14:
@RexBerry That is one of the goals in this project. We want to make it so that the code is more readable and more functional.

Byrdman32, 2022-09-15:
@RexBerry I was reminded that a project we were working on at the end of last year ended up getting rid of a lot of the conditionals that you started cleaning up yesterday. We haven't completely finished that project which is why it hasn't been merged to dev yet.

Before doing much more on this project I would advice merging `feature/gate_search` into your branch. Since the changes it will make will get rid of a fair bit of your work.

`git merge feature/gate_search`

Byrdman32, 2022-09-23:
@RexBerry Make sure when you are making commits that you reference this issue number. I have fixed the commit messages that you have already made.

autc7, 2022-09-29:
Hey guys, I'm having a bit of trouble understanding all the eedits being made here, where can I go to edit the code?

RexBerry, 2022-09-29:
@autc7 My changes are on the feature/conditionals_cleanup branch.

copper-head, 2022-10-02:
To clarify this issue, there is a new gate_approach.py file located in the feature/gate_search branch. Before any work is continued that file needs to effectively replace the gate_approach file currently used in all other branches as the new algorithm in that file was much more reliable when tested. That being said it still needs to be tested some to be prudent.

Next Steps to resolve this issue:

1. Test the algorithm in branch feature/gate_search to make sure that it is working. Try using a number of approach routes to get a solid test distribution.
2. Compare that code with the current approaching_gate.py code in dev branches and look for differences that could potentially cause issues.

Before merging to a main branch we will also need to modify the code to move the two algorithms that are computing the gps points to the algorithms folder, instead of having the a functions defined within approaching_gate.py itself. The functions that need moved are at the bottom of the approaching_gate.py file (find_gate_path and camera_to_gps_coordinate). Once the above to things have been completed I'll outline the next steps.

If you want to ask any questions, or have me go through with you and explain the code for better understanding, feel free to reach out to me on discord. My name is Duncan Truitt in RoveSoDiscord.

Byrdman32, 2022-10-02:
@copper-head the feature/gate_search branch was merged into the branch for this project.